adafruit/ArduinoCore-samd #203: Newer Adafruit BSPs cause Arduino LMIC to hang during init

terrillmoore, 2020-01-08:
I wanted to raise this issue, although I do not have much time in the next few weeks to investigate in depth (including no time to bisect-test).

Recent Adafruit SAMD BSPs are not compatible with the [MCCI Catena Arduino-LMIC](https://github.com/mcci-catena/Arduino-LMIC) sample applications -- see https://github.com/mcci-catena/arduino-lmic/issues/521.

Things work well with MCCI's BSP for the Feather M0, which is a fork of the Adafruit BSP from early 2018 -- last Adafruit commit is 69071fd0b67411fa2a3836abce6385c9892bfbec. This still worked well when @brentru added the tutorial scripts in October 2018.

Symptom: `os_init()` (the first LMIC API) never returns. There is a config table, `lmic_pinmap`, which is used by `os_init()` to set up hardware. It seems likely that some LMIC assumption about pre-conditions is no longer valid -- or maybe never was, but didn't cause problems until now.

Posting here for reference. The workaround is to use MCCI's fork of the BSP, which also adds the LoRaWAN region selection parameters to the UI.

ladyada, 2020-01-08:
git cloning' the bsp into 'hardware' will let you go from commit to commit till you find the one that 'broke' it!

terrillmoore, 2020-01-09:
Alas,,.... the tinyusb subrepo complicates this. As does the changing of bossac versions. It looks very difficult to get all the tools right across the "adafruit"/"arduino"/"tools" boundary -- the known good checkout builds, but only after I remove tinyusb (git submodule breakage); then the IDE hangs while trying to download, probably because of Bossac 1.7.0 bitrot). This looks like a problem that has to be debugged. Maybe someone from the LMIC community will have time to dig into it.

terrillmoore, 2020-01-11:
Turns out that I was in too much of a hurry, and didn't revert to exactly the fork point of our BSP. I assumed that our code didn't have a patch to make it LMIC friendly. It does. The LMIC disables interrupts during initialization and then waits for some time to pass; this is profoundly non-portable. It apparently works fine on AVRs, but not on stock M0 ports. The fix, of course, is "don't do that" in the LMIC. I think I now know enough about the code to fix that properly. Closing this.

closed 2020-01-11 by terrillmoore