adafruit/ArduinoCore-samd #204: Only 8 bytes gets received with ItsyBitsy M4 Express HID device and HID_GET_REPORT.

RaceCore24, 2020-01-10:
This problem is tested on the ItsyBitsy M4 Express.

When we send a HID_GET_REPORT to the M4 USB HID device. Only 8 bytes get received.

I created a sketch and demo application to show the problem.

[sketch.zip](https://github.com/adafruit/ArduinoCore-samd/files/4045196/sketch.zip)

Please fix this.

Thanks
 Kjetil

ladyada, 2020-01-10:
are you using the tinyusb or arduino usb core?

RaceCore24, 2020-01-10:
I am using the Arduino USB stack.

ladyada, 2020-01-10:
try tinyusb

RaceCore24, 2020-01-11:
The Arduino USB stack must work. We use it as it is easier to move from board to board.

ladyada, 2020-01-11:
please try tinyusb

RaceCore24, 2020-01-11:
We will not use tinyusb. Now we have submitted a bug in your Arduino USB core stack. I expect you to support it and fix it now that you are aware of it. Your "try tinyusb" is not helpful as our codebase does not include that support. We cannot change our entire codebase because you don't want to fix a bug.

RaceCore24, 2020-01-11:
Seriously? Are we deleting posts now? Seriously there is a bug in your arduino usb code. Why is it so difficult to understand that and fix it? Deleting posts now? I cannot just use tinyusb. It is not helpful to just spew out that. We need help and have it fixed.

ptorrone, 2020-01-11:
hi @RaceCore24 if there are bullying / abusive comments, yes, they will be removed. we removed your previous comment.

please consider this the last and final warning.

please try tinyusb at this time, that will help determine what the issue may or may not be.