adafruit/ArduinoCore-samd #208: No (or incorrect) openocd_scripts for M4 boards...

WestfW, 2020-01-27:
There is a `variants/openocd_scripts/` directory for most of the Cortex-M0 SAMD21-based boards like the Arduino Zero, but these are either missing or incorrect (copies of the `variants/arduino_zero/openocd_scripts/arduino_zero.cfg` script.)

These are apparently used by other IDEs for debugger (gdb) access to boards...
https://forums.adafruit.com/viewtopic.php?f=63&t=159048

ladyada, 2020-01-27:
openocd did not work with samd51 last we tried it - feel free to submit a PR with working scripts!

GitLang, 2021-01-10:
Has this been revisited by anybody please, or anybody working on it?.. Looking through openOCD repository there is certainly plenty of reference to and definitions for SAMD51, including the ATSAMD51P20A used on the grand Central M4. See http://openocd.org/doc/doxygen/html/atsame5_8c_source.html
It would be good if anybody with the knowledge could look into this please. I'm sure I'm not alone in spending $$$ on a Segger JLink and SWD cabling to be stuck with a brick!.

GitLang, 2021-01-11:
Further info on this. Although OpenOCD.org appear to has stopped at v0.10 a couple of years ago, it is still actively being developed at https://sourceforge.net/p/openocd/code/ci/master/tree/
On cloning it, I found "ocd\openocd-code\tcl\target\atsame5x.cfg" . It seems that SAMD51 is covered by the 5e target config file. The comment at the top of that file says  "# Microchip (former Atmel) SAM E54, E53, E51 and D51 devices
# with a Cortex-M4 core"

So the SAMD51 is clearly now supported by OpenOCD 0,40. I personally do not have the knowledge to incorporate this into Arduino, but if Adafruit have tried before, it now seems that if they try again it should work. Could Adafruit take a look at this please so those of us who have invested in a J-Link can use it with the SWD port supplied on some of your M4 boards, such as the grand central. Thank you.

ladyada, 2021-01-11:
hi, we have no ETA to look into this. every time we try this out it becomes a huge mess of patchwork code and in the end, we really just recommend getting a jlink mini for $20 as it works perfectly every time.