adafruit/ArduinoCore-samd #215: Cannot use Wire inside class

MCUdude, 2020-02-19:
Hi!
I stumbled across this issue when working on a larger project. To prevent you guys from reading through thousands of code lines, I've created a small sketch that demonstrates the problem. Not that I can compile this for an Atmel AVR, but not for a SAMD based microcontroller. In this case, the M0 Express. Why doesn't this compile? And what can I do to make it compile?

MyClassTest.ino
```c++
#include "myClass.h"

MyClass my(0x34);

void setup()
{
  my.begin();
}

void loop()
{

}
```

myClass.h
```c++
#ifndef MYCLASS_H
#define MYCLASS_H

#include <Arduino.h>
#include <Wire.h>

class MyClass
{
  public:
    MyClass(uint8_t i2cAddress);
    void begin(TwoWire &WireObject = Wire);
    
  private:
    const uint8_t _i2cAddress;
    TwoWire _WireObject;
};  

#endif
```

myClass.cpp
```c++
#include "myClass.h"

MyClass::MyClass(uint8_t i2cAddress) : _i2cAddress(i2cAddress)
{
  
}

void MyClass::begin(TwoWire &WireObject)
{
  _WireObject = WireObject;
  _WireObject.begin();
}
```

Error:
```
In file included from C:\Users\user\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.7\variants\metro_m0/variant.h:42:0,

                 from C:\Users\user\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.7\cores\arduino/delay.h:24,

                 from C:\Users\user\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.7\cores\arduino/Arduino.h:81,

                 from C:\Users\HBULL~1.LAU\AppData\Local\Temp\arduino_build_279892\sketch\myClass.h:4,

                 from C:\Users\HBULL~1.LAU\AppData\Local\Temp\arduino_build_279892\sketch\myClass.cpp:1:

C:\Users\user\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.7\cores\arduino/SERCOM.h: In member function 'void SERCOM::setClockSource(int8_t, SercomClockSource, bool)':

C:\Users\user\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.7\cores\arduino/SERCOM.h:248:30: warning: unused parameter 'idx' [-Wunused-parameter]

   void setClockSource(int8_t idx, SercomClockSource src, bool core) { };

                              ^~~

C:\Users\user\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.7\cores\arduino/SERCOM.h:248:53: warning: unused parameter 'src' [-Wunused-parameter]

   void setClockSource(int8_t idx, SercomClockSource src, bool core) { };

                                                     ^~~

C:\Users\user\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.7\cores\arduino/SERCOM.h:248:63: warning: unused parameter 'core' [-Wunused-parameter]

   void setClockSource(int8_t idx, SercomClockSource src, bool core) { };

                                                               ^~~~

In file included from C:\Users\user\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.7\variants\metro_m0/variant.h:42:0,

                 from C:\Users\user\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.7\cores\arduino/delay.h:24,

                 from C:\Users\user\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.7\cores\arduino/Arduino.h:81,

                 from C:\Users\HBULL~1.LAU\AppData\Local\Temp\arduino_build_279892\sketch\MyClassTest.ino.cpp:1:

C:\Users\user\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.7\cores\arduino/SERCOM.h: In member function 'void SERCOM::setClockSource(int8_t, SercomClockSource, bool)':

C:\Users\user\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.7\cores\arduino/SERCOM.h:248:30: warning: unused parameter 'idx' [-Wunused-parameter]

   void setClockSource(int8_t idx, SercomClockSource src, bool core) { };

                              ^~~

C:\Users\user\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.7\cores\arduino/SERCOM.h:248:53: warning: unused parameter 'src' [-Wunused-parameter]

   void setClockSource(int8_t idx, SercomClockSource src, bool core) { };

                                                     ^~~

C:\Users\user\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.7\cores\arduino/SERCOM.h:248:63: warning: unused parameter 'core' [-Wunused-parameter]

   void setClockSource(int8_t idx, SercomClockSource src, bool core) { };

                                                               ^~~~

C:\Users\HBULL~1.LAU\AppData\Local\Temp\arduino_build_279892\sketch\myClass.cpp: In constructor 'MyClass::MyClass(uint8_t)':

myClass.cpp:3:62: error: no matching function for call to 'TwoWire::TwoWire()'

 MyClass::MyClass(uint8_t i2cAddress) : _i2cAddress(i2cAddress)

                                                              ^

In file included from C:\Users\HBULL~1.LAU\AppData\Local\Temp\arduino_build_279892\sketch\myClass.h:5:0,

                 from C:\Users\HBULL~1.LAU\AppData\Local\Temp\arduino_build_279892\sketch\myClass.cpp:1:

C:\Users\user\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.7\libraries\Wire/Wire.h:35:5: note: candidate: TwoWire::TwoWire(SERCOM*, uint8_t, uint8_t)

     TwoWire(SERCOM *s, uint8_t pinSDA, uint8_t pinSCL);

     ^~~~~~~

C:\Users\user\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.7\libraries\Wire/Wire.h:35:5: note:   candidate expects 3 arguments, 0 provided

C:\Users\user\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.7\libraries\Wire/Wire.h:32:7: note: candidate: constexpr TwoWire::TwoWire(const TwoWire&)

 class TwoWire : public Stream

       ^~~~~~~

C:\Users\user\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.7\libraries\Wire/Wire.h:32:7: note:   candidate expects 1 argument, 0 provided

C:\Users\user\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.7\libraries\Wire/Wire.h:32:7: note: candidate: constexpr TwoWire::TwoWire(TwoWire&&)

C:\Users\user\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.7\libraries\Wire/Wire.h:32:7: note:   candidate expects 1 argument, 0 provided

Using library Wire at version 1.0 in folder: C:\Users\user\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.7\libraries\Wire 
exit status 1
no matching function for call to 'TwoWire::TwoWire()'
```

ladyada, 2020-02-19:
hi please use pointers to Wire rather than pass by reference :)

closed 2020-02-19 by ladyada

MCUdude, 2020-02-19:
Cool, will try this. Would you mind explaining why this matters? :)

ladyada, 2020-02-19:
not sure tbh...but we pass by pointer all the time and it works :D

MCUdude, 2020-02-19:
> not sure tbh...but we pass by pointer all the time and it works :D

Well, that's good enough for me!

MCUdude, 2020-02-19:
@MagnusRasmussenGit here's the solution!