adafruit/ArduinoCore-samd #216: Warnings 1.5.10 everytime the library update dialog pops up on the CI_Tests folder

fab672000, 2020-02-26:
We now get a repeated warning:
`CI_Tests: no headers files (.h) found in C:\Users\MyUserName\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.10\libraries\CI_Tests
Invalid library found in C:\Users\<MyUserName>\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.5.10\libraries\CI_Tests`

A simple workaround is to simply delete this testing folder (containing only one very interesting fast fft test example btw) after installing update.

Finally, shouldn't  this example reusing the cmsis lib deserve in the examples section for the SAMD based boards ?

ladyada, 2020-02-26:
@hathach you can add an empty header file to get rid of the error

hathach, 2020-02-26:
This is already fixed by #214 , should be fixed when 1.5.11 is released.

commit 647e7296 (2017-04-21): Removed wrong cast to uint8

Fix #216