adafruit/ArduinoCore-samd #240: -On option of the SAMD21 board.

dojyorin, 2020-07-22:
I want to select optimization (-On) in the compile option of the SAMD21 board.

The sketch I write for SAMD21 has plenty of free space.
I want to use that free space for speed up.

Would you consider adding the "-On" option to compilation settings?
Thank you.

hathach, 2020-07-28:
we won't  add an extra menu to do this. You could try to edit `platform.txt` or bettter `platform.local.txt` to do so
https://arduino.github.io/arduino-cli/platform-specification/#platformlocaltxt

closed 2020-07-28 by hathach

dojyorin, 2020-07-31:
> we won't add an extra menu just for this. You could try to edit `platform.txt` or bettter `platform.local.txt` to do so
> https://arduino.github.io/arduino-cli/platform-specification/#platformlocaltxt

OK, I'll try.
Thanks😊