adafruit/ArduinoCore-samd #245: Serial.printf of uint64_t

trlafleur, 2020-08-06:
Repost of 244...

As you can see, %llu is NOT supported...

If I try to use Serial.printf to print two uint64_t variables, the 2nd one prints as a zero...

Arduino 1.8.13
Feather M0
Adafruit SAMD21 Boards 1.6.0

moved from (arduino/ArduinoCore-sam#109)
~~~
Output with %lu
hello...
A: 12345678, B; 0
A: 12345678
B: 87654321

Output with %llu
hello...
A: lu, B; lu
A: lu
B: lu
~~~

~~~
void setup() {
     Serial.begin (115200);              // Initialize USB/serial connection
     delay(2000);

     Serial.printf("hello...\n");

uint64_t A = 12345678l;
uint64_t B = 87654321l;

Serial.printf ("A: %lu, B; %lu\n", A, B);

Serial.printf ("A: %lu\n", A);
Serial.printf ("B: %lu\n", B);
   
}

void loop() {
  // put your main code here, to run repeatedly:

}
~~~

~~~
void setup() {
     Serial.begin (115200);              // Initialize USB/serial connection
     delay(2000);

     Serial.printf("hello...\n");

uint64_t A = 12345678ll;
uint64_t B = 87654321ll;

Serial.printf ("A: %llu, B; %llu\n", A, B);

Serial.printf ("A: %llu\n", A);
Serial.printf ("B: %llu\n", B);
   
}

void loop() {
  // put your main code here, to run repeatedly:

}
~~~

hathach, 2020-08-06:
again please post this on the support forum.

closed 2020-08-06 by hathach

trlafleur, 2020-08-06:
It’s bug... should be listed on github issues...

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

hathach, 2020-08-06:
I did a quick check, the implementation use std newlib `vsnprintf()`, if it doesn't not work, it is not supported. You could try to change newlib nano link option etc.. to get it work. Though we don't do it ourself, that's why I asked you to open topic on forum support, there are more people there.

trlafleur, 2020-08-06:
I understand that %llu may not be supported, put trying to print two
numbers on the same printf statement should work and is a bug...

I would suggest that you leave this open and I will also post on forum...

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

hathach, 2020-08-07:
prtinf with uint64_t is not supported, the behavior is undefined, it is not a bug.