adafruit/ArduinoCore-samd #248: Build warnings -- cleanup fallthrough warnings

henrygab, 2020-08-14:
Currently three files use the following to disable this warning on a couple functions:
```C++
#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wimplicit-fallthrough"
// ... code that generates warnings ...
#pragma GCC diagnostic pop
```

That was done as a temporary measure in #207 to allow finding/fixing other issues in the code exposed by compiling with `-Wall`.

Here's a list of correct ways to fix these implicit fallthrough warnings:
* [ ]  For C++17 and higher, use [`[[fallthrough]]`](https://en.cppreference.com/w/cpp/language/attributes/fallthrough) attribute
* [ ]  For GCC at C++11 / C++14, use [`[[gnu::fallthrough]]`](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wimplicit-fallthrough_003d)
* [ ]  For CLANG at C++11 / C++14, use [`[[clang::fallthrough]]`](http://clang.llvm.org/docs/LanguageExtensions.html#feature-checking-macros) attribute
* [ ]  For GCC with older compilers, use [`__attribute__(( fallthrough ))`](https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Wimplicit-fallthrough_003d)

NOTE: The comment method is legacy and very fragile. See: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=78497

If code will be compiled by multiple compilers / versions, consider defining a macro such as ADAFRUIT_FALLTHROUGH, that conditionally defines to C++17, GCC, or CLANG as available.

hathach, 2020-08-24:
there is no need for `ADAFRUIT_FALLTHROUGH` just either use attribute fallthru or  [[fallthrough]]. We only need to fix current gcc and worry about other compiler later on when needed.

hathach, 2020-08-27:
I really like __attribute__(( fallthrough )); It should be able to work with existing old & new code, just use it in another library, works like a charm

https://github.com/adafruit/Adafruit_SPIFlash/pull/70/files#diff-0d585c34005499efb9fdee8db227e2eaR111

henrygab, 2020-08-27:
I agree that is a preferred attribute.  Now the difficulty is in verifying that the code is actually doing the "**right thing**" in each case, before using this attribute.  Else, the attribute will hide a true bug from smart compiler warnings.

hathach, 2020-08-31:
@henrygab yeah, but I don't think we could find anyone that could insert it into the "right place" any time soon :(. Let's leave it as it is for now :)