adafruit/ArduinoCore-samd #253: Build warnings -- [-Wrestrict]

henrygab, 2020-08-14:
```
/home/runner/.arduino15/packages/adafruit/hardware/samd/1.6.0/cores/arduino/WString.cpp: In member function 'void String::remove(unsigned int, unsigned int)':
/home/runner/.arduino15/packages/adafruit/hardware/samd/1.6.0/cores/arduino/WString.cpp:701:9: warning: 'char* strncpy(char*, const char*, size_t)' accessing 0 or more bytes at offsets [-1073741824, 1073741823] and [-1073741824, 1073741823] may overlap up to 4294967295 bytes at offset [5368709117, 1073741823] [-Wrestrict]
  701 |  strncpy(writeTo, buffer + index + count,len - index);
      |  ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

See example logs at https://github.com/adafruit/ArduinoCore-samd/runs/965558867

hathach, 2020-08-25:
Hi @henrygab you could just submit PR directly without filling these issue, you don't even have to explain the necessity of this -W option. We could just enable and get rid of them as many as we could. I don't want you to spend more time than needed on these issues, just submitted the PR, paste the compiler output ( I skip the English text anyway :smile: )

henrygab, 2020-09-05:
I took a look at the code.  It's actually a simple fix here, as the code is 100% of the time providing overlapped pointers to `strncpy()`.  :
<details><summary>The code is removing a prefix from the string</summary><P/>

https://github.com/adafruit/ArduinoCore-samd/blob/8d3b49bf3926dd0353c97657e84cecf106b5ea0d/cores/arduino/WString.cpp#L695-L703

</details>

Therefore, by the time `strncpy()` is called, there's no error condition possible if re-implementing this to simply copy characters one at a time (a simple `for` loop).

Unfortunately, I only have this BSP locally via board manager at the moment.  But, this is an ultra-simple fix.

PR #289 (2021-04-13): Fix [-Wrestrict] bug
Should fix #253, where the issue is well documented.

**_WARNING: ENTIRELY UNTESTED (although trivial) CODE._**
closed 2021-04-13 by hathach