adafruit/ArduinoCore-samd #255: Tone() not working on SAMD51

RAyWB, 2020-08-19:
Just tried to use Tone() to generate a simple square signal, but nothing happens.
tried several boards , itsybitsyM4 , GrandCentral ,MetroM4 ,same situation.

think it should work just using:

tone(9,1000,20000);
notone();
to generate 1 KHz square signal on pin9 for 20 seconds.

can you please check what´s wrong?

ladyada, 2020-08-19:
hi can you try older versions of the board support package, and tell us wher eit stopped working? that would help figure out what happened that made it stop :)

RAyWB, 2020-08-19:
Hi , it stopped working on change from 1.5.3 to 1.5.4
1.5.3 works

ladyada, 2020-08-19:
ok thanks @hathach can you maybe look?

RAyWB, 2020-08-19:
seems not to be related on tone.cpp , copying tone.cpp from 1.5.3  to newer version doesn´t help.
sorry for no more ideas as I´m not a pro in coding.

hathach, 2020-08-20:
Thanks for reporting the issue, I will try to check this out as soon as I could

hathach, 2020-08-27:
I confirmed the issue on SAMD51, it even seem to cause the MCU to be hanged as well somehow. Looking into it now

RAyWB, 2020-08-28:
Thanks for taking care of this issue

hathach, 2020-08-28:
@RAyWB PR #257 should fix this issue, please try it out and let's us know your result.

RAyWB, 2020-08-28:
Tone() checked on GrandCentral and ItsyBitsyM4 , works !
Thanks for fixing that issue

PR #257 (2020-08-28): Fix tone on SAMD51
commit 1e9a669f07cce881c2861dfbc330cc0c5a0115d3 tries to use TC0 for tone, and TC1 for Servo. However, the servo  remain unchanged (keep using TC0) causing the resource conflict. This PR does
- change servo to use TC1 in respect of the above commit intention
- Update Tone.cpp to match upstream version https://github.com/arduino/ArduinoCore-samd/commits/master/cores/arduino/Tone.cpp which received a few fix since we last added.
- Make define `Tone_Handler() as  TC0_Handler` so that it is a strong symbol. The issue with current weak alias is that we have multiple weak symbol f [... cut]
closed 2020-08-28 by hathach