adafruit/ArduinoCore-samd #266: Bootloader folder "grand_central_m4" missing from the ArduinoCore-samd

DCelectronics, 2020-09-27:
I need to the bootloader file "bootloader-grandcentralM4-v2.0.0-adafruit.5.bin" or "bootloader-grandcentralM4-v2.0.0-adafruit.5.elf" because I want to experiment with the Grand Central M4 board and Atmel Studio.
After I erase the ATSAMD51P20A chip, I want to be able to restore the Grand Central M4 board to its original condition with Atmel Studio.

hathach, 2020-09-30:
thanks, we will add the missing bootloader file soon, you could find the latest binary of bootlaoder here in its repo's release assests. 

https://github.com/adafruit/uf2-samdx1/releases

DCelectronics, 2020-09-30:
Thank for your response.

- It addresses the need to have a .bin bootloader file to program a fully erased ATSAMD51P20A chip on Grand Central M4 board.

- It provides the link to latest version of the bootloader and eliminates the extra step to drag & drop the "update-bootloader-grandcentral_m4-v3.10.0.uf2" file in the CGM4BOOT storage media after flashing the chip with the requested "bootloader-grandcentral_m4-v2.0.0-adafruit.9.bin" file.

Would you please consider including the source files as well?

Best Regards!

dhalbert, 2020-09-30:
You need to set the BOOTPROT fuses to protect the bootloader if you load it manually. I am not sure if your process The update-bootloader .uf2 does this as part of the updating process, which is why we recommend using it. If you erase the bootloader completely you could load it manually and then run the update-bootloader uf2 for the same version to do this easily.

hathach, 2020-10-01:
> Would you please consider including the source files as well?

which source files you are mentioning, if you mean the bootlaode source, it is there in the above repo.

DCelectronics, 2020-10-01:
> 
> 
> > Would you please consider including the source files as well?
> 
> which source files you are mentioning, if you mean the bootlaode source, it is there in the above repo.

Here is an example of the Metro M4 board which includes the bootloader source files.
https://github.com/adafruit/ArduinoCore-samd/tree/master/bootloaders/metroM4
It's nice to have the source files and make the bootloader from scratch as a learning opportunity and start experimenting with it.
Your response addressed the need to locate a bootloader .bin file for the Grand Central M4 board.
I should revise the title of this issue to "Missing Grand Central M4 bootloader .bin file" and close it.
Thanks!

**The back story.**
What prompted me to requesting the "bootloader-grandcentralM4-v2.0.0-adafruit.5.bin" file was when I tried to program the Grand Central M4 bootloader with the Arduino IDE and gave me this error message, which pointed at the location of the needed bootloader.

*Arduino: 1.8.13 (Windows 10), Board: "Adafruit Grand Central M4 (SAMD51), Enabled, 120 MHz (standard), Small (-Os) (standard), 50 MHz (standard), Arduino, Off"
C:\Users\user\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7/bin/openocd.exe -d2 -s C:\Users\user\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7/share/openocd/scripts/ -f C:\Users\user\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.6.2/variants/grand_central_m4/openocd_scripts/arduino_zero.cfg -c telnet_port disabled; init; halt; at91samd bootloader 0; program {C:\Users\user\AppData\Local\Arduino15\packages\adafruit\hardware\samd\***1.6.2/bootloaders/grand_central_m4/bootloader-grandcentralM4-v2.0.0-adafruit.5.bin***} verify reset; shutdown
Open On-Chip Debugger 0.10.0+dev-gf0767a31 (2018-06-11-13:36)
...blah blah
embedded:startup.tcl:60: Error: Can't find C:\Users\user\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.6.2/variants/grand_central_m4/openocd_scripts/arduino_zero.cfg
in procedure 'script'
at file "embedded:startup.tcl", line 60
Error while burning bootloader.*

I have successfully programmed the bootloader for Metro M0 and Feather M0 with Arduino IDE (with previously unlocked and fully erased chips) and locked the fuse BOOTPROT afterwards with Atmel Studio 7. 
Currently,this is not possible (with Arduino IDE) for some of the M4 boards (Feather M4 and Metro M4) I tried out because they are missing the "openocd_scripts/xxxxxxx_m0.cfg" files.

For example, the Metro M0 has the openocd_scripts folder with the "metro_m0.cfg" file.
ArduinoCore-samd/variants/metro_m0/openocd_scripts/ metro_m0.cfg
Howerver, the following boards do not have the "openocd_scripts" folder.
ArduinoCore-samd/variants/feather_m4/
ArduinoCore-samd/variants/grand_central_m4/
ArduinoCore-samd/variants/metro_m4/

However, this a separate issue, which would be resolved at some time as the parts of the project get committed.

closed 2020-10-01 by ladyada