adafruit/ArduinoCore-samd #271: Bug: Internal Compiler Error

arock093, 2020-11-05:
Raspberry Pi Zero W
Board: Adafruit Trinket M0

I'm trying to compile and getting an internal compiler error.

arduino-cli compile --fqbn adafruit:samd:adafruit_trinket_m0 MyFirstSketch
/home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.3/cores/arduino/WInterrupts.c: In function 'detachInterrupt':
/home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.3/cores/arduino/WInterrupts.c:378:1: internal compiler error: Illegal instruction
}
^
Please submit a full bug report,
with preprocessed source if appropriate.
See https://gcc.gnu.org/bugs/ for instructions.
/home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.3/cores/arduino/cortex_handlers.c: In function 'Reset_Handler':
/home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.3/cores/arduino/cortex_handlers.c:543:1: internal compiler error: Illegal instruction
}

I am able to compile for a different board.
arduino-cli compile --fqbn arduino:avr:mega MyFirstSketch
Sketch uses 1536 bytes (0%) of program storage space. Maximum is 253952 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 8183 bytes for local variables. Maximum is 8192 bytes.

------

gcc --version
gcc (Raspbian 8.3.0-6+rpi1) 8.3.0

Section of verbose output:
/home/pi/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/arm-none-eabi-gcc -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -MMD "-D__SKETCH_NAME__="""MyFirstSketch.ino"""" -DF_CPU=48000000L -DARDUINO=10607 -DARDUINO_TRINKET_M0 -DARDUINO_ARCH_SAMD -DCRYSTALLESS -DADAFRUIT_TRINKET_M0 -D__SAMD21E18A__ -DARM_MATH_CM0PLUS -DUSB_VID=0x239A -DUSB_PID=0x801E -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER="Adafruit"" "-DUSB_PRODUCT="Trinket M0"" -I/home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.4/cores/arduino/TinyUSB -I/home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.4/cores/arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore -I/home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.4/cores/arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src -Os -I/home/pi/.arduino15/packages/adafruit/tools/CMSIS/5.4.0/CMSIS/Core/Include/ -I/home/pi/.arduino15/packages/adafruit/tools/CMSIS/5.4.0/CMSIS/DSP/Include/ -I/home/pi/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/ -I/home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.4/cores/arduino -I/home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.4/variants/trinket_m0 /home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.4/cores/arduino/WInterrupts.c -o /tmp/arduino-sketch-704E54BF84B4765302A16BBD33C1F4E6/core/WInterrupts.c.o
/home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.4/cores/arduino/WInterrupts.c: In function 'detachInterrupt':
/home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.4/cores/arduino/WInterrupts.c:378:1: internal compiler error: Illegal instruction
}
^
Please submit a full bug report,
with preprocessed source if appropriate.
See https://gcc.gnu.org/bugs/ for instructions.
/home/pi/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/arm-none-eabi-gcc -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -MMD "-D__SKETCH_NAME__="""MyFirstSketch.ino"""" -DF_CPU=48000000L -DARDUINO=10607 -DARDUINO_TRINKET_M0 -DARDUINO_ARCH_SAMD -DCRYSTALLESS -DADAFRUIT_TRINKET_M0 -D__SAMD21E18A__ -DARM_MATH_CM0PLUS -DUSB_VID=0x239A -DUSB_PID=0x801E -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER="Adafruit"" "-DUSB_PRODUCT="Trinket M0"" -I/home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.4/cores/arduino/TinyUSB -I/home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.4/cores/arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore -I/home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.4/cores/arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src -Os -I/home/pi/.arduino15/packages/adafruit/tools/CMSIS/5.4.0/CMSIS/Core/Include/ -I/home/pi/.arduino15/packages/adafruit/tools/CMSIS/5.4.0/CMSIS/DSP/Include/ -I/home/pi/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0/CMSIS/Device/ATMEL/ -I/home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.4/cores/arduino -I/home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.4/variants/trinket_m0 /home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.4/cores/arduino/cortex_handlers.c -o /tmp/arduino-sketch-704E54BF84B4765302A16BBD33C1F4E6/core/cortex_handlers.c.o
/home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.4/cores/arduino/cortex_handlers.c: In function 'Reset_Handler':
/home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.4/cores/arduino/cortex_handlers.c:545:1: internal compiler error: Illegal instruction
}
^
Please submit a full bug report,
with preprocessed source if appropriate.
See https://gcc.gnu.org/bugs/ for instructions.
Error during build: exit status 1

I was thinking about manually updating gcc but I'm not sure if it's using Raspberry PI OS included gcc to compile or the one it downloads with Adafruit:samd.

oaoutpost0, 2021-01-12:
I get a near identical error when trying to compile a bare minimum example for the Feather M0 on an RPi 1. Since arock093's error is on an RPi Zero, I suspect the issue is related to the armv6 instruction set. I use an RPi 1 for development since it shares the same CPU architecture as the Pi zero and has more ports. I can compile identical code for the same board from my Windows machine without any issues. The error must be a problem in the GCC toolchain.

In the meantime, I'm going to try an older release of the samd boards to see if the older toolchain does work on arm 6.

```
Arduino: 1.8.13 (Linux), Board: "Adafruit Feather M0, Small (-Os) (standard), Arduino, Off"

/home/pi/Documents/arduino-1.8.13/arduino-builder -dump-prefs -logger=machine -hardware /home/pi/Documents/arduino-1.8.13/hardware -hardware /home/pi/.arduino15/packages -tools /home/pi/Documents/arduino-1.8.13/tools-builder -tools /home/pi/Documents/arduino-1.8.13/hardware/tools/avr -tools /home/pi/.arduino15/packages -built-in-libraries /home/pi/Documents/arduino-1.8.13/libraries -libraries /home/pi/Arduino/libraries -fqbn=adafruit:samd:adafruit_feather_m0:opt=small,usbstack=arduino,debug=off -vid-pid=239A_800B -ide-version=10813 -build-path /tmp/arduino_build_123648 -warnings=default -build-cache /tmp/arduino_cache_555705 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=/home/pi/.arduino15/packages/arduino/tools/bossac/1.7.0-arduino3 -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=/home/pi/.arduino15/packages/arduino/tools/bossac/1.8.0-48-gb176eee -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/pi/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4 -prefs=runtime.tools.arm-none-eabi-gcc-9-2019q4.path=/home/pi/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4 -prefs=runtime.tools.arduinoOTA.path=/home/pi/.arduino15/packages/arduino/tools/arduinoOTA/1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=/home/pi/.arduino15/packages/arduino/tools/arduinoOTA/1.2.1 -prefs=runtime.tools.CMSIS.path=/home/pi/.arduino15/packages/adafruit/tools/CMSIS/5.4.0 -prefs=runtime.tools.CMSIS-5.4.0.path=/home/pi/.arduino15/packages/adafruit/tools/CMSIS/5.4.0 -prefs=runtime.tools.CMSIS-Atmel.path=/home/pi/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=/home/pi/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0 -prefs=runtime.tools.openocd.path=/home/pi/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=/home/pi/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -prefs=runtime.tools.bossac-1.7.0-arduino3.path=/home/pi/.arduino15/packages/arduino/tools/bossac/1.7.0-arduino3 -verbose /home/pi/Documents/arduino-1.8.13/examples/01.Basics/BareMinimum/BareMinimum.ino
/home/pi/Documents/arduino-1.8.13/arduino-builder -compile -logger=machine -hardware /home/pi/Documents/arduino-1.8.13/hardware -hardware /home/pi/.arduino15/packages -tools /home/pi/Documents/arduino-1.8.13/tools-builder -tools /home/pi/Documents/arduino-1.8.13/hardware/tools/avr -tools /home/pi/.arduino15/packages -built-in-libraries /home/pi/Documents/arduino-1.8.13/libraries -libraries /home/pi/Arduino/libraries -fqbn=adafruit:samd:adafruit_feather_m0:opt=small,usbstack=arduino,debug=off -vid-pid=239A_800B -ide-version=10813 -build-path /tmp/arduino_build_123648 -warnings=default -build-cache /tmp/arduino_cache_555705 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=/home/pi/.arduino15/packages/arduino/tools/bossac/1.7.0-arduino3 -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=/home/pi/.arduino15/packages/arduino/tools/bossac/1.8.0-48-gb176eee -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/pi/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4 -prefs=runtime.tools.arm-none-eabi-gcc-9-2019q4.path=/home/pi/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4 -prefs=runtime.tools.arduinoOTA.path=/home/pi/.arduino15/packages/arduino/tools/arduinoOTA/1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=/home/pi/.arduino15/packages/arduino/tools/arduinoOTA/1.2.1 -prefs=runtime.tools.CMSIS.path=/home/pi/.arduino15/packages/adafruit/tools/CMSIS/5.4.0 -prefs=runtime.tools.CMSIS-5.4.0.path=/home/pi/.arduino15/packages/adafruit/tools/CMSIS/5.4.0 -prefs=runtime.tools.CMSIS-Atmel.path=/home/pi/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=/home/pi/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0 -prefs=runtime.tools.openocd.path=/home/pi/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=/home/pi/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -prefs=runtime.tools.bossac-1.7.0-arduino3.path=/home/pi/.arduino15/packages/arduino/tools/bossac/1.7.0-arduino3 -verbose /home/pi/Documents/arduino-1.8.13/examples/01.Basics/BareMinimum/BareMinimum.ino
Using board 'adafruit_feather_m0' from platform in folder: /home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.4
Using core 'arduino' from platform in folder: /home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.4
Detecting libraries used...
/home/pi/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/arm-none-eabi-g++ -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions "-D__SKETCH_NAME__=\"\"\"BareM [... cut ...] duino_build_123648/core/cortex_handlers.c.o
/home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.4/cores/arduino/cortex_handlers.c: In function 'Reset_Handler':
/home/pi/.arduino15/packages/adafruit/hardware/samd/1.6.4/cores/arduino/cortex_handlers.c:545:1: internal compiler error: Illegal instruction
 }
 ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
exit status 1
Error compiling for board Adafruit Feather M0.
```

oaoutpost0, 2021-01-12:
And I can confirm that version 4.8.3-2014q1 of arm-none-eabi can compile the code just fine (release 1.4.1). Of course, while it compiles fine, it encounters a file not found error for {runtime.tools.bossac-1.7.0.path}/bossac. Hooray for progress!

oaoutpost0, 2021-01-12:
Fixed both issues by reinstalling the SAMD boards. I used version 1.6.21 for the Arduino SAMD boards and 1.4.1 for the Adafruit SAMD boards.