adafruit/ArduinoCore-samd #272: Analog Read Issue

Daniel314159, 2020-11-16:
I am reading voltage between 3.3v and 0v on A2 on the M4 Grandcentral. The board seems to randomly lock up due to firmware issue at specific voltages. I am running the board @ 12bit resolution and have tried 11bit resolution. I will be trying 10bit yet but would prefer to use 12bit for my application. 

I am using VScode/Platformio and C to program the board. The error is below:

`Exception in thread rx:
Traceback (most recent call last):
  File "C:\Users\Daniel\.platformio\python3\lib\threading.py", line 926, in _bootstrap_inner
    self.run()
  File "C:\Users\Daniel\.platformio\python3\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "c:\users\daniel\.platformio\penv\lib\site-packages\serial\tools\miniterm.py", line 445, in reader
    data = self.serial.read(self.serial.in_waiting or 1)
  File "c:\users\daniel\.platformio\penv\lib\site-packages\serial\serialwin32.py", line 257, in in_waiting
    raise SerialException("ClearCommError failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: ClearCommError failed (PermissionError(13, 'The device does not recognize the command.', None, 22))`

The code is here: 
` static bool cal = 0; //switch used to avoid locking in calibration loop.
  static long trun_target_position = 0;
  long open_position = 81400;      // fully open in theory = 81514;
  float voltage = 0;               //input raw voltage
  static float voltage_memory = 0; //holds previous measured voltage

  voltage = analogRead(input_voltage);
  for (int i = 0; i < 1000; i++)
  {
    voltage = voltage + analogRead(input_voltage);
    smart_delay(3);
  }
  voltage = voltage / 1000;
  Serial.println("Voltage");
  Serial.println(voltage);`

ladyada, 2020-11-16:
please post a complete but minimal Arduino sketch demonstrating the issue

closed 2020-11-24 by Daniel314159

Daniel314159, 2020-11-24:
I apologize for not submitting a code sample. I was pressed for time and ended up using a different board.