adafruit/ArduinoCore-samd #273: Access to old releases

PTS93, 2020-12-01:
I noticed that there are only releases listed with version 1.6.0 and older.
Though I do know there were about 5-6 releases prior to that. One of them being 1.10 which I'm looking for.
Are those archived somewhere? Even the git tags don't seem to list those any more.

PTS93, 2020-12-01:
It seems there was maybe just a rename of tags?
These very old releases (which are from the time frame of the 1.10 release) seem out of order. There are f.e. two 1.6.1 releases. One very old one and one quite recent one.
[image]

PTS93, 2020-12-01:
It seems they are still available via the Arduino Board Manager. That is enough for me atm but maybe someone wants to check why the releases naming in git seems to be a bit out of order :)

closed 2020-12-01 by PTS93

ladyada, 2020-12-01:
so...technically, the git releases are not used by the arduino manager, it is handled differently, its just for our personal enjoyment
we wont be going back to previous versions to 'retag' them - it will end in tears.

PTS93, 2020-12-01:
I see! They seemed to somewhat correlate so it looked like something got messed up by accident :)