adafruit/ArduinoCore-samd #282: Feather M0 Proto not working with AirLift Featherwing

DrNebin, 2021-01-29:
The ScanTest code pin definitions do not allow the correct pins to be assigned for the Feather M0 Proto. The code must be edited by hand to assign the correct pins. You can see a thread related to this here:

https://forums.adafruit.com/viewtopic.php?f=53&t=174743&p=851663#p851663

I am not sure if the boards.txt file needs updated (I think maybe it does not) or if the sample code with the library simply needs to be edited to include the Feather M0 Proto as one of the defined boards for pin assignments.

brentru, 2021-01-29:
This is an issue within the WiFiNina examples, not this repository. I'll open a PR there to resolve the issue.

The define `defined(ADAFRUIT_FEATHER_M0_EXPRESS)` is only for the M0 Express, the M0 Proto should be defined in the examples as `defined(ADAFRUIT_FEATHER_M0)`

brentru, 2021-01-29:
Merged in https://github.com/adafruit/WiFiNINA/pull/9#pullrequestreview-579664465

closed 2021-01-29 by brentru