adafruit/ArduinoCore-samd #292: Update of Arduino Servo library for SAMD21 and SAMD51

agrommek, 2021-04-13:
Hi everybody,

I added some code to the Servo library for better support of SAMD21 and SAMD51 board. The library now supports 12 servos per timer and up to 24 servos total, for both SAMD21 and SAMD51. Tested on Feather M0 and Feather M4. I would like to create a pull request for that, but I am not sure to which repository. 

My patch would resolve  [an issue at the repository for the Arduino Servo library](https://github.com/arduino-libraries/Servo/issues/21), opened by @Silvanosky in 2019. There is already a long-pending-but-not-yet-merged [pull request](https://github.com/arduino-libraries/Servo/pull/34) by @ladyada which fixes part of the problem. Because the pull request was not merged, Adafruit duplicated the Servo library here at Adafruit's fork of the ArduinoCore-samd repository with the modifications. I didn't realize at first that there was an open issue upstream, so I based my code off of the version of the Servo library in Adafruit's ArduinoCore-samd repository.

Now, what would be the best way to get the changed code into the upstream Ardino Servo library?

- Would Adafruit care to accept a pull requenst to the local version of the Servo library in the ArduinoCore-samd repository and update the pull request upstream?
- Or should I directly make a pull request upstream, with Adafruit's pull reqeust being rejected?

Before I forget it, here is the [link to my code](https://github.com/agrommek/ArduinoCore-samd/tree/fix-servo-1/libraries/Servo/src/samd). I would be grateful for   :-)

ladyada, 2021-04-13:
our servo library works pretty well for samd51 and samd21, we'd probably not take a refactor to our code unless its fixing a serious bug. a small, concise PR would be best.

feel free to submit a PR to upstream. we have no insight as to whether it would be merged

closed 2021-04-13 by ladyada

agrommek, 2021-04-13:
Hi @ladyada 

Thank you for your prompt response. I will try to submit a PR upstream. 

However, just for the record: There _is_ a serious bug in in servo library for the SAMD chips, although it is not in your code, but in the original Arduino SAMD servo code. This bug cannot be fixed without some refactoring. 

Just try to use 12 servos (as advertised by the library) with an average pulse width of more than 1820 microseconds (i.e. more than about 21.85 ms total pulse duration for all servos taken together) on a Feather M0 or Arduino Zero. The pulse widths will be ok, but the PWM period/frequency for the servos will go from 20 ms/50 Hz to more than 35 ms/less than 29 Hz. This can be clearly observed on an oscilloscope. Many servos will not work with this kind of timing, I think. The behavior is due to the overflow of the TC counter value after about 21.85 ms. 

I will open a new issue upstream ASAP (i.e. tomorrow or so).

ladyada, 2021-04-13:
oh yeah we understand what you're talking about. i wonder why we dont turn all the pulses on at once, then turn them off at different times. maybe theres more jitter?

agrommek, 2021-04-13:
I _think_ I know why "one does it like this". To me, this appears to be a legacy thing from radio-controlled cars/planes/etc. with analog electronics. There, one _has_ to time-multiplex different servos (i.e. put the pulses back-to-back) in order to be able to control up to 6 servos with a single radio channel. Why only 6, and not up to 10? There would be room for up to ten 2-ms-pulses in a 20-ms-period, after all. Because radio regulations say, that you are not allowed to transmit nonstop, otherwise the generated sidebands would influence neighboring radio channels (at least [this PDF](http://www.princeton.edu/~mae412/TEXT/NTRAK2002/292-302.pdf) mentions something along those lines).

These radio-remote-control-arguments are, of course, not applicable to micorcontrollers. But old habits appear to die hard... :smile: