adafruit/ArduinoCore-samd #294: Arduino Board Manager: "Warning: non trusted contribution"

timonsku, 2021-04-20:
Updating my Arduino core yields this error, which sounds like it would potentially leave an unfinished installation:

`Warning: non trusted contribution, skipping script execution (C:\Users\Timon\AppData\Local\Arduino15\packages\adafruit\hardware\samd\1.6.6\post_install.bat)`

hathach, 2021-05-04:
yeah, seems like it is only to install CDC driver. Maybe Arduino IDE change how thing is allowed to be executed. Though it should not affect (much) Adafruit repo. It seems mostly for Arduino SAMD board (as this is a fork of that). Adafruit boards has its own driver installation. Thanks for raising the issue, we could leave it open to see if there is any user has troubles with it.

per1234, 2022-02-18:
The [post-install script](https://arduino.github.io/arduino-cli/latest/platform-specification/#post-install-script) restriction has been removed as unnecessary:
https://github.com/arduino/arduino-cli/pull/911

The change has not yet propagated to Arduino IDE 1.x, so the script will still not run during the "Adafruit SAMD Boards" installation for users of Arduino IDE 1.8.19 and older, but it will run when they are using Arduino IDE 2.x or Arduino CLI directly.