adafruit/ArduinoCore-samd #297: Compilation error with Feather M4 CAN: `CANSAME5x.cpp:12:10: fatal error: same51.h: No such file or directory`

gromain, 2021-05-08:
- Arduino board:  Feather M4 CAN

- Arduino IDE version:  1.8.13

- Steps to reproduce the problem:

1. Install Arduino SAMD boards, version 1.8.11
1. Install Adafruit SAMD boards, version 1.6.7
2. Install library CAN Adafruit Fork from Library Manager, version 1.1.0
3. Choose board Adafruit Feather M4 CAN (SAME51)
4. Open example CANSender
5. Compile

The error message displayed is quite clear: `Arduino/libraries/CAN_Adafruit_Fork/src/CANSAME5x.cpp:12:10: fatal error: same51.h: No such file or directory`

```
Arduino: 1.8.13 (Linux), Board: "Adafruit Feather M4 CAN (SAME51), Enabled, 120 MHz (standard), Small (-Os) (standard), 50 MHz (standard), Arduino, Off"

/usr/share/arduino/arduino-builder -dump-prefs -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/rbazile/.arduino15/packages -hardware /home/rbazile/Documents/codes/Arduino/hardware -tools /usr/share/arduino/tools-builder -tools /home/rbazile/.arduino15/packages -libraries /home/rbazile/Documents/codes/Arduino/libraries -fqbn=adafruit:samd:adafruit_feather_m4_can:cache=on,speed=120,opt=small,maxqspi=50,usbstack=arduino,debug=off -vid-pid=239A_80CD -ide-version=10813 -build-path /tmp/arduino_build_860810 -warnings=default -build-cache /tmp/arduino_cache_523296 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=/home/rbazile/.arduino15/packages/arduino/tools/bossac/1.7.0-arduino3 -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=/home/rbazile/.arduino15/packages/arduino/tools/bossac/1.8.0-48-gb176eee -prefs=runtime.tools.bossac-1.7.0-arduino3.path=/home/rbazile/.arduino15/packages/arduino/tools/bossac/1.7.0-arduino3 -prefs=runtime.tools.arduinoOTA.path=/home/rbazile/.arduino15/packages/arduino/tools/arduinoOTA/1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=/home/rbazile/.arduino15/packages/arduino/tools/arduinoOTA/1.2.1 -prefs=runtime.tools.CMSIS.path=/home/rbazile/.arduino15/packages/adafruit/tools/CMSIS/5.4.0 -prefs=runtime.tools.CMSIS-5.4.0.path=/home/rbazile/.arduino15/packages/adafruit/tools/CMSIS/5.4.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/rbazile/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4 -prefs=runtime.tools.arm-none-eabi-gcc-9-2019q4.path=/home/rbazile/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4 -prefs=runtime.tools.CMSIS-Atmel.path=/home/rbazile/.arduino15/packages/adafruit/tools/CMSIS-Atmel/1.2.1 -prefs=runtime.tools.CMSIS-Atmel-1.2.1.path=/home/rbazile/.arduino15/packages/adafruit/tools/CMSIS-Atmel/1.2.1 -prefs=runtime.tools.openocd.path=/home/rbazile/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=/home/rbazile/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -verbose /home/rbazile/Documents/codes/Arduino/libraries/CAN_Adafruit_Fork/examples/CANSender/CANSender.ino
/usr/share/arduino/arduino-builder -compile -logger=machine -hardware /usr/share/arduino/hardware -hardware /home/rbazile/.arduino15/packages -hardware /home/rbazile/Documents/codes/Arduino/hardware -tools /usr/share/arduino/tools-builder -tools /home/rbazile/.arduino15/packages -libraries /home/rbazile/Documents/codes/Arduino/libraries -fqbn=adafruit:samd:adafruit_feather_m4_can:cache=on,speed=120,opt=small,maxqspi=50,usbstack=arduino,debug=off -vid-pid=239A_80CD -ide-version=10813 -build-path /tmp/arduino_build_860810 -warnings=default -build-cache /tmp/arduino_cache_523296 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=/home/rbazile/.arduino15/packages/arduino/tools/bossac/1.7.0-arduino3 -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=/home/rbazile/.arduino15/packages/arduino/tools/bossac/1.8.0-48-gb176eee -prefs=runtime.tools.bossac-1.7.0-arduino3.path=/home/rbazile/.arduino15/packages/arduino/tools/bossac/1.7.0-arduino3 -prefs=runtime.tools.arduinoOTA.path=/home/rbazile/.arduino15/packages/arduino/tools/arduinoOTA/1.2.1 -prefs=runtime.tools.arduinoOTA-1.2.1.path=/home/rbazile/.arduino15/packages/arduino/tools/arduinoOTA/1.2.1 -prefs=runtime.tools.CMSIS.path=/home/rbazile/.arduino15/packages/adafruit/tools/CMSIS/5.4.0 -prefs=runtime.tools.CMSIS-5.4.0.path=/home/rbazile/.arduino15/packages/adafruit/tools/CMSIS/5.4.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=/home/rbazile/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4 -prefs=runtime.tools.arm-none-eabi-gcc-9-2019q4.path=/home/rbazile/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4 -prefs=runtime.tools.CMSIS-Atmel.path=/home/rbazile/.arduino15/packages/adafruit/tools/CMSIS-Atmel/1.2.1 -prefs=runtime.tools.CMSIS-Atmel-1.2.1.path=/home/rbazile/.arduino15/packages/adafruit/tools/CMSIS-Atmel/1.2.1 -prefs=runtime.tools.openocd.path=/home/rbazile/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -prefs=runtime.tools.openocd-0.10.0-arduino7.path=/home/rbazile/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino7 -verbose /home/rbazile/Documents/codes/Arduino/libraries/CAN_Adafruit_Fork/examples/CANSender/CANSender.ino
Using board 'adafruit_feather_m4_can' from platform in folder: /home/rbazile/.arduino15/packages/adafruit/hardware/samd/1.6.7
Using core 'arduino' from platform in folder: /home/rbazile/.arduino15/packages/adafruit/hardware/samd/1.6.7
Detecting libraries used...
/home/rbazile/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/arm-none-eabi-g++ -mcpu=cortex-m4 -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions "-D__SKETCH_NAME__=\"\"\"CANSender.ino\"\"\"" -w -x c++ -E -CC -DF_CPU=120000000L -DARDUINO=10813 -DARDUINO_FEATHER_M4_CAN -DARDUINO_ARCH_SAMD -D__SAME51J19A__ -DADAFRUIT_FEATHER_M4_EXPRESS -DADAFRUIT_FEATHER_M4_CAN -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x80CD -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit LLC\"" "-DUSB_PRODUCT=\"Adafruit Feather M4 CAN\"" -I/home/rbazile/.arduino15/packages/adafruit/hardware/samd/1.6.7/cores/arduino/TinyUSB -I/home/rbazile/.arduino15/packages/adafruit/hardware/samd/1.6.7/cores/arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore -I/home/rbazile/.arduino15/packages/adafruit/hardware/samd/1.6.7/cores/arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DENABLE_CACHE -Os -DVARIANT_QSPI_BAUD_DEFAULT=50000000 -D__SAME51J19A__ -DADAFRUIT_FEATHER_M4_EXPRESS -DADAFRUIT_FEATHER_M4_CAN -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x80CD -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit LLC\"" "-DUSB_PRODUCT=\"Adafruit Feather M4 CAN\"" -I/home/rbazile/.arduino15/packages/adafruit/hardware/samd/1.6.7/cores/arduino/TinyUSB -I/home/rbazile/.arduino15/packages/adafruit/hardware/samd/1.6.7/cores/arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore -I/home/rbazile/.arduino15/packages/adafruit/hardware/samd/1.6.7/cores/arduino/TinyUSB/Adafruit_TinyUSB_ArduinoCore/tinyusb/src -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -I/home/rbazile/.arduino15/packages/adafruit/tools/CMSIS/5.4.0/CMSIS/Core/Include/ -I/home/rbazile/.arduino15/packages/adafruit/tools/CMSIS/5.4.0/CMSIS/DSP/Include/ -I/home/rbazile/.arduino15/packages/adafruit/tools/CMSIS-Atmel/1.2.1/CMSIS/Device/ATMEL/ -I/home/rbazile/.arduino15/packages/adafruit/hardware/samd/1.6.7/cores/arduino -I/home/rbazile/.arduino15/packages/adafruit/hardware/samd/1.6.7/variants/feather_m4_can /tmp/arduino_build_860810/sketch/CANSender.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for CAN.h: [CAN_Adafruit_Fork@1.1.0]
ResolveLibrary(CAN.h)
  -> candidates: [CAN_Adafruit_Fork@1.1.0]
/home/rbazile/.arduino15/packages/adafruit/tools/arm-none-eabi-gcc/9-2019q4/bin/arm-none-eabi-g++ -mcpu=cortex-m4 -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions "-D__SKETCH_NAME__=\"\"\"CANSender.ino\"\"\"" -w -x c++ -E -CC -DF_CPU=120000000L -DARDUINO=10813 -DARDUINO_FEATHER_M4_CAN -DARDUINO_ARCH_SAMD -D__SAME51J19A__ -DADAFRUIT_FEATHER_M4_EXPRESS -DADAFRUIT_FEATHER_M4_CAN -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x80CD -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit LLC\"" "-DUSB_PRODUCT=\"Adafruit Feather M4 CAN\"" -I/home/rbazile/.arduino15/packages/adafruit/hardware/samd/1.6.7/cores/arduino/TinyUSB -I/home/rbazile/.arduino15/packages/adafruit/hardware/samd/1.6.7/cores/arduino/TinyUSB/Adafruit_TinyUSB_ArduinoC [... cut ...]
Alternatives for same51.h: []
/home/rbazile/Documents/codes/Arduino/libraries/CAN_Adafruit_Fork/src/CANSAME5x.cpp:12:10: fatal error: same51.h: No such file or directory
ResolveLibrary(same51.h)
  -> candidates: []
   12 | #include "same51.h"
      |          ^~~~~~~~~~
compilation terminated.
Using library CAN_Adafruit_Fork at version 1.1.0 in folder: /home/rbazile/Documents/codes/Arduino/libraries/CAN_Adafruit_Fork 
exit status 1
Error compiling for board Adafruit Feather M4 CAN (SAME51).
```

ladyada, 2021-05-08:
@hathach  can you take a look

@gromain you can use a previous release of adafruit samd until it is fixed

hathach, 2021-05-10:
This issue will be fixed by https://github.com/adafruit/arduino-CAN/pull/5 . The #291 put the same51.h as part of sam.h which is included by default when Arduino.h is include. I.e we don't ever need to manually include same51.h. It is just a left over bug from the migration. Thank you for your issue, once PR is merged, we will release the library shortly afterwards.

gromain, 2021-05-10:
Awesome! Thanks a lot!

closed 2021-05-10 by ladyada