adafruit/ArduinoCore-samd #304: analogWrite on Atmel SAMD behaves differently than described by official ASrduino documentation

pauljurczak, 2021-06-27:
I’m working with Atmel SAMD microcontroller on Adafruit Feather M0 board. Looking at the documentation here: https://www.arduino.cc/reference/en/language/functions/analog-io/analogwrite/, I read that PWM will be stopped or changed by next call to analogWrite() (or a call to digitalRead() or digitalWrite()) on the same pin. In my tests, only analogWrite() affects the PWM on given pin, e.g.:
```
analogWrite(1, 128);
analogWrite(1, 0);
```
will leave pin 1 in low state, but
```
analogWrite(1, 128);
digitalWrite(1, 0);
```
will leave it producing 50% duty cycle PWM. Forum post https://community.platformio.org/t/analogwrite-on-atmel-samd-behaves-differently-than-documented/22202 has a detailed explanation why it's happening.

ladyada, 2021-06-27:
could be! want to submit a PR to tweak the behavior of digitalwrite to disable pwm output?

pauljurczak, 2021-06-27:
Sure, why not. Before I do, are there any performance reasons, which would make adding an extra conditional instruction
```
if (timer != NOT_ON_TIMER) turnOffPWM(timer);
```
not acceptable, e.g. bit banging on I/O port?

ladyada, 2021-06-27:
arduino isnt very performant - if it works on a variety of tests thats ok!