adafruit/ArduinoCore-samd #305: how to detect adafruit boards at compile time?

2bndy5, 2021-07-14:
I need a way to detect if the SAMD core being used if from this fork (not the arduino/ArduinoCore-samd repo). The reason for this demand is because our library makes use of the `Serial.printf()` in this fork, however it is not present in the arduino/ArduinoUnified-API repo (as used by the arduino/ArduinoCore-samd repo).

Initially by using 
```cpp
            #if defined (ARDUINO_ARCH_SAMD) && !defined (ARDUINO_API_VERSION)
                // likely using the adafruit/ArduinoCore-samd repo
                #define printf_P Serial.printf
            #endif // defined (ARDUINO_ARCH_SAMD)
```
I thought that I could single out any board supported by this fork, but I'm now getting a user report that would indicate that it also gets triggered for the Arduino MKR WIFI 1010 board (which doesn't have `Serial.printf()`). I know I could just do a `defined (ADAFRUIT_FEATHER_M0)` but I don't want to do this for the 30 or so boards this fork supports. Is there a better way?

hathach, 2021-07-14:
yeah, there is no explicit way to detect if the board is using this fork vs arduino one. Maybe we could add macro `ARDUINO_SAMD_ADAFRUIT` to the compile flag similar to what we did with nrf52. @ladyada do you have any better name or suggestion.

ladyada, 2021-07-14:
`ADAFRUIT_FEATHER_M0` ?

https://github.com/adafruit/ArduinoCore-samd/blob/master/boards.txt#L46

2bndy5, 2021-07-14:
Yes please. That would be most convenient and future proof.

ladyada, 2021-07-14:
@2bndy5 ok please try it out and make a PR!

PR #306 (2021-07-15): add -DADAFRUIT_ARCH_SAMD to all board's build.extra_flags
resolves #305 

I chose the macro ADAFRUIT_ARCH_SAMD as it seems most like using ARDUINO_ARCH_SAMD macro.

tested with 
```cpp
            #if defined (ADAFRUIT_ARCH_SAMD)
                // using the adafruit/ArduinoCore-samd repo
                #define printf_P Serial.printf
            #endif // defined (ADAFRUIT_ARCH_SAMD)
```
verified working on QT PY M0. I don't have any Arduino produced SAMD-based boards, but the compiler output (targetting the MKR WIFI 1010 board) doesn't show the `-DADAFRUIT_ARCH_SAMD` being passed as an extra [... cut]
closed 2021-07-15 by hathach