adafruit/ArduinoCore-samd #320: Add defines for internal DotStar pins

caternuson, 2022-01-26:
For boards that have onboard (internal) DotStars:

- Adafruit Trinket M0
- Itsy Bitsy M0 Express
- Adafruit PyRuler
- FunHouse
- Adafruit ItsyBitsy nRF52840
- Adafruit GEMMA M0
- Adafruit Itsybitsy m4
- others?

add `#def`s for the DotStar pins, ex:
```cpp
// DotStar LED
#define INTERNAL_DS_DATA     3
#define INTERNAL_DS_CLK      4
```
to make referencing them easy in code.

hathach, 2022-01-26:
Is it OK to name it as `PIN_DOTSTAR_DATA, PIN_DOTSTAR_CLK, DOTSTAR_NUM` to be consistent with existing neopixel naming. I could help to implement this based on schematic 

https://github.com/adafruit/ArduinoCore-samd/blob/2a7aae79ee52ba95ce70ec25a551e3a60d2ce5f6/variants/circuitplay/variant.h#L94-L95

ladyada, 2022-01-26:
yep!

caternuson, 2022-01-26:
Thanks. Yes, let's decide name now. Easy to change to whatever at this point.

For boards that already have something, like Gemma - replace existing? or just add new defs?

caternuson, 2022-01-28:
Done with #321

closed 2022-01-28 by caternuson