adafruit/ArduinoCore-samd #322: Update the TinyUSB library?

Spegs21, 2022-01-28:
Can we update the TinyUSB library included in the core? The newer revs have TinyUSBDevice.setSerialDescriptor() which I need for my TinyUSB implementation here:  Legion2/CorsairLightingProtocol#215

I'm happy to create the PR if needed.

hathach, 2022-01-29:
we will do it gradually, BSP has longer release cycle than library. Meanwhile you could just install TinyUSB from library manager, it will be picked by arduino instead of the one in the core.

Spegs21, 2022-01-29:
@hathach I reverted the core version on my local install and installed the library from library manager and it worked as you described. I thought I tried this earlier and it didn't work but it must have been an issue with my implementation. Thanks for looking into it.

closed 2022-01-29 by Spegs21