adafruit/ArduinoCore-samd #329: Arduino PWM pin conflicts on SAMD21

kattni, 2022-05-10:
From user feedback:
> I am pretty sure there are some PWM pin conflicts on QT Py, for example, pin 4 is mapped to TCC0_CH6, and pin 10 is mapped to TCC0_CH2, but for this board variant, there's a line of code that says `TCC0_CH6 = (0<<8)|(2), // Channel 6 is 2!`.
> 
> I've observed this in practice. If both pins are running PWM, and you change the duty cycle of one, the other one changes with it.
> ```
> pinMode(4, OUTPUT);
> pinMode(10, OUTPUT);
> analogWrite(10, 1);
> analogWrite(4, 128);
> ```
> Measure the output on pin 10, the duty cycle is 50%, when it should be nearly zero.
> 
> It's not something that can be fixed, it's nobody's fault, but it'll be nice if the tutorial page for the QT Py mentioned this.

The variant mentioned is in core Arduino, and is from the original Arduino code. It might be a hardware limitation. The question is should we document it in the SAMD21 board guides? Or is this an unusual and unlikely situation to encounter?

@dhalbert Found these links and thought they might somehow be related. 
https://forum.arduino.cc/t/samd21-wo-and-cc-register-mapping/852449
https://github.com/arduino/ArduinoCore-samd/pull/170

hathach, 2022-05-19:
Just take a look at the SAMD21 manual, TCC0 only has 4 channels, and CH6 in indeed CH2, likewise ch4,5,6,7 --> ch0, 1, 2, 3. The reason seems to allow 1 signal to drive multiple pins synchronously e.g fading of 2 LEDs, or maybe having other usage. So this is definitely the hardware limitation.

ladyada, 2022-05-21:
@hathach could you try to adjust the definition so that the main timer for pins is:

- A2 TCC0[0]
- A3 TCC0[1]
- SDA TCC2[0]
- SCL TCC2[1]
- TX TCC1[0]
- RX TCC1[1]
- SCK TCC0[3] (CHANGED to make unique!)
- MISO TCC1[3] (CHANGED to be unique!)
- MOSI TCC0[2]  (CHANGED to be unique!)

this would make each pin on a different TCC?

hathach, 2022-05-26:
> @hathach could you try to adjust the definition so that the main timer for pins is:
>
> * A2 TCC0[0]
> * A3 TCC0[1]
> * SDA TCC2[0]
> * SCL TCC2[1]
> * TX TCC1[0]
> * RX TCC1[1]
> * SCK TCC0[3] (CHANGED to make [...]

This is spot on, I should have figured it out. #331 is submitted.

PR #331 (2022-06-05): change the SDA, SCL timer from alt to primary to avoid pwm conflict
- fix pwm conflict on qtpy, close #329 
- also correct the PA02 which is not on timer 

@kattni
closed 2022-06-05 by ladyada