adafruit/ArduinoCore-samd #330: `adafruit:samd:adafruit_feather_m4` Cannot use `--build-propery` flag with Arduino CLI

zfields, 2022-05-18:
If I compile my sketch without specifying the `--build-property`, then it compiles correctly:
```
$ arduino-cli compile --fqbn=adafruit:samd:adafruit_feather_m4 --warnings all ./examples/Example1_NotecardBasics/Example1_NotecardBasics.ino
Sketch uses 26592 bytes (5%) of program storage space. Maximum is 507904 bytes.
```

When I add the `--build-property`, then it goes sideways:
```
$ arduino-cli compile --build-property build.extra_flags=-Werror --fqbn=adafruit:samd:adafruit_feather_m4 --warnings all ./examples/Example1_NotecardBasics/Example1_NotecardBasics.ino
In file included from /root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/Arduino.h:65,
                 from /tmp/arduino-sketch-2136EA99E4E95C26D41A391DD84A8876/sketch/Example1_NotecardBasics.ino.cpp:1:
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/WVariant.h:257:36: error: 'TCC_INST_NUM' was not declared in this scope
  257 | extern const void* g_apTCInstances[TCC_INST_NUM+TC_INST_NUM] ;
      |                                    ^~~~~~~~~~~~
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/WVariant.h:257:49: error: 'TC_INST_NUM' was not declared in this scope
  257 | extern const void* g_apTCInstances[TCC_INST_NUM+TC_INST_NUM] ;
      |                                                 ^~~~~~~~~~~
In file included from /root/.arduino15/packages/adafruit/hardware/samd/1.7.10/variants/feather_m4/variant.h:46,
                 from /root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/delay.h:24,
                 from /root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/Arduino.h:80,
                 from /tmp/arduino-sketch-2136EA99E4E95C26D41A391DD84A8876/sketch/Example1_NotecardBasics.ino.cpp:1:
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/SERCOM.h:171:16: error: expected ')' before '*' token
  171 |   SERCOM(Sercom* s) ;
      |         ~      ^
      |                )
In file included from /root/.arduino15/packages/adafruit/hardware/samd/1.7.10/variants/feather_m4/variant.h:46,
                 from /root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/delay.h:24,
                 from /root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/Arduino.h:80,
                 from /tmp/arduino-sketch-2136EA99E4E95C26D41A391DD84A8876/sketch/Example1_NotecardBasics.ino.cpp:1:
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/SERCOM.h:257:3: error: 'Sercom' does not name a type
  257 |   Sercom* sercom;
      |   ^~~~~~
In file included from /root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/USB/USBAPI.h:37,
                 from /root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/Arduino.h:158,
                 from /tmp/arduino-sketch-2136EA99E4E95C26D41A391DD84A8876/sketch/Example1_NotecardBasics.ino.cpp:1:
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/USB/SAMD21_USBDevice.h:175:2: error: 'UsbDevice' does not name a type
  175 |  UsbDevice &usb;
      |  ^~~~~~~~~
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/USB/SAMD21_USBDevice.h:178:34: error: 'UsbDeviceDescriptor' does not name a type; did you mean 'DeviceDescriptor'?
  178 |  __attribute__((__aligned__(4))) UsbDeviceDescriptor EP[USB_EPT_NUM];
      |                                  ^~~~~~~~~~~~~~~~~~~
      |                                  DeviceDescriptor
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/USB/SAMD21_USBDevice.h: In constructor 'USBDevice_SAMD21G18x::USBDevice_SAMD21G18x()':
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/USB/SAMD21_USBDevice.h:32:27: error: class 'USBDevice_SAMD21G18x' does not have any field named 'usb'
   32 |  USBDevice_SAMD21G18x() : usb(USB->DEVICE) {
      |                           ^~~
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/USB/SAMD21_USBDevice.h:32:31: error: 'USB' was not declared in this scope
   32 |  USBDevice_SAMD21G18x() : usb(USB->DEVICE) {
      |                               ^~~
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/USB/SAMD21_USBDevice.h: In member function 'void USBDevice_SAMD21G18x::enable()':
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/USB/SAMD21_USBDevice.h:44:3: error: 'usb' was not declared in this scope
   44 |   usb.CTRLA.bit.ENABLE = 1;
      |   ^~~
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/USB/SAMD21_USBDevice.h: In member function 'void USBDevice_SAMD21G18x::disable()':
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/USB/SAMD21_USBDevice.h:50:3: error: 'usb' was not declared in this scope
   50 |   usb.CTRLA.bit.ENABLE = 0;
      |   ^~~
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/USB/SAMD21_USBDevice.h: In member function 'void USBDevice_SAMD21G18x::setUSBDeviceMode()':
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/USB/SAMD21_USBDevice.h:57:35: error: 'usb' was not declared in this scope
   57 |  inline void setUSBDeviceMode() { usb.CTRLA.bit.MODE = USB_CTRLA_MODE_DEVICE_Val; }
      |                                   ^~~
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/USB/SAMD21_USBDevice.h:57:56: error: 'USB_CTRLA_MODE_DEVICE_Val' was not declared in this scope
   57 |  inline void setUSBDeviceMode() { usb.CTRLA.bit.MODE = USB_CTRLA_MODE_DEVICE_Val; }
      |                                                        ^~~~~~~~~~~~~~~~~~~~~~~~~
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/USB/SAMD21_USBDevice.h: In member function 'void USBDevice_SAMD21G18x::setUSBHostMode()':
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/USB/SAMD21_USBDevice.h:58:35: error: 'usb' was not declared in this scope
   58 |  inline void setUSBHostMode()   { usb.CTRLA.bit.MODE = USB_CTRLA_MODE_HOST_Val;   }
      |                                   ^~~
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/USB/SAMD21_USBDevice.h:58:56: error: 'USB_CTRLA_MODE_HOST_Val' was not declared in this scope
   58 |  inline void setUSBHostMode()   { usb.CTRLA.bit.MODE = USB_CTRLA_MODE_HOST_Val;   }
[... 317 lines cut ...]
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/USB/SAMD21_USBDevice.h:188:22: error: '__get_PRIMASK' was not declared in this scope
  188 |  __Guard() : primask(__get_PRIMASK()), loops(1) {
      |                      ^~~~~~~~~~~~~
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/USB/SAMD21_USBDevice.h:189:3: error: '__disable_irq' was not declared in this scope
  189 |   __disable_irq();
      |   ^~~~~~~~~~~~~
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/USB/SAMD21_USBDevice.h: In destructor '__Guard::~__Guard()':
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/USB/SAMD21_USBDevice.h:193:4: error: '__enable_irq' was not declared in this scope
  193 |    __enable_irq();
      |    ^~~~~~~~~~~~
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/USB/SAMD21_USBDevice.h:195:4: error: '__ISB' was not declared in this scope
  195 |    __ISB();
      |    ^~~~~
In file included from /root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/USB/USB_host.h:23,
                 from /root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/Arduino.h:159,
                 from /tmp/arduino-sketch-2136EA99E4E95C26D41A391DD84A8876/sketch/Example1_NotecardBasics.ino.cpp:1:
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/USB/samd21_host.h: At global scope:
/root/.arduino15/packages/adafruit/hardware/samd/1.7.10/cores/arduino/USB/samd21_host.h:26:49: error: 'UsbHostDescriptor' does not name a type
   26 | extern __attribute__((__aligned__(4))) volatile UsbHostDescriptor usb_pipe_table[USB_EPT_NUM];
      |                                                 ^~~~~~~~~~~~~~~~~

Error during build: exit status 1
```

This is a [known issue](https://github.com/arduino/arduino-cli/issues/846) with the Arduino CLI. Some vendors, like ST for example, provide [workarounds like `build_opt.h`](https://github.com/stm32duino/wiki/wiki/Customize-build-options-using-build_opt.h). Does Adafruit support such a workaround, or is there anyway to pass addition flags to the compiler?

hathach, 2022-05-19:
@zfields please edit your first post to have compile log as attached txt for readability.

zfields, 2022-05-20:
FWIW, `--build-propery.extra_flags` is available to all the standard Arduino platforms. Apparently, whether it's poor documentation or not, the general public seems to be gravitating toward this flag.

> _@per1234 I'm not suggesting that it is a "defect", but rather a known point of confusion for Arduino CLI users. I'll add my thoughts regarding this issue to the linked Arduino CLI issue to keep from polluting this thread._

zfields, 2022-05-21:
I have validated the build-property `compiler.cpp.extra_flags` (suggested by @per1234) does work with the `adafruit:samd:adafruit_feather_m4`.

As stated by @per1234 `build.extra_flags` is reserved for the platform vendor (e.g. Adafruit), while `compiler.cpp.extra_flags` is to be used by the end-user.

closed 2022-05-21 by zfields

ladyada, 2022-05-21:
thanks for the quick resolution @zfields  and help @per1234 !