adafruit/ArduinoCore-samd #332: Unable to build MetroM4 Bootloader: "error: 'PORT' undeclared"

ricardoboss, 2022-06-28:
I am trying to compile the [metroM4 bootloader](https://github.com/adafruit/ArduinoCore-samd/tree/master/bootloaders/metroM4) using make for Windows. I ran into multiple problems, like the arduino libraries not being present, which I was able to fix using arduino-cli. I also tried to set up and build everything in WSL, but I was not successful.

I installed the following arduino cores using:
```
arduino-cli core install arduino:sam
arduino-cli core install arduino:samd
```

This is the output of `make.exe`:

<details>
<summary>output with original versions</summary>

```
----------------------------------------------------------
Compiling bootloader using
BASE PATH = C:\Users\rboss\AppData\Local/Arduino15/packages/arduino
GCC  PATH = C:\Users\rboss\AppData\Local/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-
----------------------------------------------------------
Creating build folder
mkdir build
----------------------------------------------------------
Compiling board_driver_led.c to build/board_driver_led.o
"C:\Users\rboss\AppData\Local/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-gcc" -mthumb -mcpu=cortex-m4 -c -std=gnu99 -MD -MP -MF "build/board_driver_led.d" -MT"build/board_driver_led.d" -MT"build/board_driver_l
ed.o" -ffunction-sections -fdata-sections -nostdlib -nostartfiles --param max-inline-insns-single=500 -Os -DDEBUG=0 -D -DBOARD_ID_ -I"C:\Users\rboss\AppData\Local/Arduino15/packages/arduino/tools/CMSIS/4.5.0/CMSIS/Include/" -I"C:\Users\rboss\AppData\Local/Arduino15/packages/arduino/tools/CMSIS-Atmel/1.1.0/CMSIS/Device/ATMEL/" board_driver_led.c -o build/board_driver_led.o
<command-line>:0:1: error: macro names must be identifiers
In file included from board_driver_led.c:20:0:
board_driver_led.h: In function 'LED_init':
board_driver_led.h:27:30: error: 'PORT' undeclared (first use in this function)
 inline void LED_init(void) { PORT->Group[BOARD_LED_PORT].DIRSET.reg = (1<<BOARD_LED_PIN); }
                              ^
board_driver_led.h:27:30: note: each undeclared identifier is reported only once for each function it appears in
board_driver_led.h: In function 'LED_on':
board_driver_led.h:28:28: error: 'PORT' undeclared (first use in this function)
 inline void LED_on(void) { PORT->Group[BOARD_LED_PORT].OUTSET.reg = (1<<BOARD_LED_PIN); }
                            ^
board_driver_led.h: In function 'LED_off':
board_driver_led.h:29:29: error: 'PORT' undeclared (first use in this function)
 inline void LED_off(void) { PORT->Group[BOARD_LED_PORT].OUTCLR.reg = (1<<BOARD_LED_PIN); }
                             ^
board_driver_led.h: In function 'LED_toggle':
board_driver_led.h:30:32: error: 'PORT' undeclared (first use in this function)
 inline void LED_toggle(void) { PORT->Group[BOARD_LED_PORT].OUTTGL.reg = (1<<BOARD_LED_PIN); }
                                ^
board_driver_led.h: In function 'LEDRX_init':
board_driver_led.h:40:32: error: 'PORT' undeclared (first use in this function)
 inline void LEDRX_init(void) { PORT->Group[BOARD_LEDRX_PORT].DIRSET.reg = (1<<BOARD_LEDRX_PIN); }
                                ^
board_driver_led.h: In function 'LEDRX_on':
board_driver_led.h:41:30: error: 'PORT' undeclared (first use in this function)
 inline void LEDRX_on(void) { PORT->Group[BOARD_LEDRX_PORT].OUTCLR.reg = (1<<BOARD_LEDRX_PIN); }
                              ^
board_driver_led.h: In function 'LEDRX_off':
board_driver_led.h:42:31: error: 'PORT' undeclared (first use in this function)
 inline void LEDRX_off(void) { PORT->Group[BOARD_LEDRX_PORT].OUTSET.reg = (1<<BOARD_LEDRX_PIN); }
                               ^
board_driver_led.h: In function 'LEDRX_toggle':
board_driver_led.h:43:34: error: 'PORT' undeclared (first use in this function)
 inline void LEDRX_toggle(void) { PORT->Group[BOARD_LEDRX_PORT].OUTTGL.reg = (1<<BOARD_LEDRX_PIN); }
                                  ^
board_driver_led.h: In function 'LEDTX_init':
board_driver_led.h:52:32: error: 'PORT' undeclared (first use in this function)
 inline void LEDTX_init(void) { PORT->Group[BOARD_LEDTX_PORT].DIRSET.reg = (1<<BOARD_LEDTX_PIN); }
                                ^
board_driver_led.h: In function 'LEDTX_on':
board_driver_led.h:53:30: error: 'PORT' undeclared (first use in this function)
 inline void LEDTX_on(void) { PORT->Group[BOARD_LEDTX_PORT].OUTCLR.reg = (1<<BOARD_LEDTX_PIN); }
                              ^
board_driver_led.h: In function 'LEDTX_off':
board_driver_led.h:54:31: error: 'PORT' undeclared (first use in this function)
[... 82 lines cut ...]
                              ^~~~
board_driver_led.h: In function 'LEDTX_off':
board_driver_led.h:54:31: error: 'PORT' undeclared (first use in this function)
 inline void LEDTX_off(void) { PORT->Group[BOARD_LEDTX_PORT].OUTSET.reg = (1<<BOARD_LEDTX_PIN); }
                               ^~~~
board_driver_led.h: In function 'LEDTX_toggle':
board_driver_led.h:55:34: error: 'PORT' undeclared (first use in this function)
 inline void LEDTX_toggle(void) { PORT->Group[BOARD_LEDTX_PORT].OUTTGL.reg = (1<<BOARD_LEDTX_PIN); }
                                  ^~~~
board_driver_led.c: At top level:
board_driver_led.c:22:10: error: unknown type name 'uint8_t'
 volatile uint8_t ledKeepValue = 0;
          ^~~~~~~
board_driver_led.c:23:10: error: unknown type name 'uint8_t'
 volatile uint8_t ledTargetValue = 20;
          ^~~~~~~
board_driver_led.c:24:10: error: unknown type name 'int8_t'
 volatile int8_t ledDirection = 1;
          ^~~~~~
make: *** [build/board_driver_led.o] Fehler 1
```

</details>

I don't have much experience with C in general, so any help or a tip to point me to the right direction would be greatly appreciated.

ricardoboss, 2022-06-28:
It looks like some libraries are missing, but which ones? And how do I get them? Do I need to configure something differently so the compiler finds them?

hathach, 2022-06-28:
the bootloader actually has its own repo here https://github.com/adafruit/uf2-samdx1 and compiled there. You should try to follow its instruction. The one within this repo is pretty much obsolete.

Note: it is always more trouble to compile on windows than linux/mac. Maybe you should try its WSL.

closed 2022-06-28 by hathach

ricardoboss, 2022-06-28:
> I also tried to set up and build everything in WSL, but I was not successful.

But thanks anyways! I will look into that repo instead.