adafruit/ArduinoCore-samd #345: Arduino IDE 2 Debugger Not Available for Feather M0 or Feather M0 Express

mjs513, 2023-06-16:
Arduino IDE 2 has built in debugger support but it does not seem to be available for any of the M0 feather boards.  A couple of us (@kurtE) got it working for different processor and we would like to try it with the M0.   Any suggestions on how to set it up or if there are any plans to implement it on these boards.

Thanks
Mike

KurtE, 2023-06-17:
Quick update and background information:

So far it looks like only the Arduino Zero has debug support within IDE2 without additional hardware.  Unfortunately I don't have one of those to try out to see how well they work.   We noticed that you stopped carrying that board and instead sell your metro M0 (https://www.adafruit.com/product/3505) which mentions:
`USB native support, comes with USB bootloader and serial port debugging
`
So wondering if it might work the same way?  But don't have one of these either ;) 
But I do have one of your Feather M0 (https://www.adafruit.com/product/3178)
And it has the same comments. So kept fingers crossed and tried it.

First attempt, built simple sketch and found that the debug button within Arduino IDE 2 was not enabled.  Knowing from the other boards we knew it needed a debug section in your platform.txt so added:

```
# Debugger configuration (general options)
# ----------------------------------------
# EXPERIMENTAL feature:
#  - this is alpha and may be subject to change without notice
debug.executable={build.path}/{build.project_name}.elf
debug.toolchain=gcc
debug.toolchain.path={runtime.tools.arm-none-eabi-gcc.path}/bin/
debug.toolchain.prefix=arm-none-eabi-
debug.server=openocd
debug.server.openocd.path={runtime.tools.openocd-0.10.0-arduino7.path}/bin/openocd
debug.server.openocd.scripts_dir={runtime.tools.openocd-0.10.0-arduino7.path}/share/openocd/scripts/
debug.server.openocd.script={runtime.platform.path}/variants/{build.variant}/{build.openocdscript}
```
Which was from the Arduino SAMD platform.txt.  I updated the paths to match other paths in your file.
Cleared out the IDE cache... and then tried it.

This time when I selected the board, the debug button was enabled.  I downloaded a sketch and then tried debug.  And it failed with some error like no transports defined or some such error. 

So I went to where the config file was located: "C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.7.12/variants/feather_m0/openocd_scripts/feather_m0.cfg"

And then compared it to the Arduino ZERO cfg file arduino_zero.cfg in their variants directory. and found they were almost identical, except the zero one has one additional line at the start of it:

`source [find interface/cmsis-dap.cfg]
`

So I added that to yours.  Not sure if this would be the right one?  But worth a try.

This time when I hit the debug command it looks like it tried a few extra things before it failed:

```
Waiting for gdb server to start...[2023-06-17T12:47:43.175Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
"C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\openocd\\0.10.0-arduino7/bin/openocd" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "c:\\Users\\kurte\\Documents\\Arduino\\Blink_hello" -f "C:/arduino-ide_nightly/resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl" -f "C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.7.12/variants/feather_m0/openocd_scripts/feather_m0.cfg"
Open On-Chip Debugger 0.10.0+dev-gf0767a31 (2018-06-11-13:36)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
none separate
adapter speed: 400 kHz
cortex_m reset_config sysresetreq
Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Error: unable to find CMSIS-DAP device

[2023-06-17T12:47:43.287Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...

```

Not sure what to try next, as I don't do much with these boards nor have had any experience with the Arduino Debug setup.

Suggestions? 

Next up Probably try a different board soldering up wires to the LINK pins.

KurtE, 2023-06-18:
Quick update:  As I mentioned on the forum: https://forums.adafruit.com/viewtopic.php?p=976274#p976274
I was able to start a debug session on a Feather M0 using an RPI Pico programmed to be a CMIS-DAP adapter connected up to the SWDIO SWCLK pins. 

But again this requires some/all of the changes I mentioned in the previous post. 

I believe those changes were in the catchup PR:  https://github.com/adafruit/ArduinoCore-samd/pull/183/files/4a743043fe92c398cbff48a357455451213f73a1..84c09b3265e2a8b548a29b141f0c9281b1baf154
which I am guessing is unlikely to be merged as it was created back in 2019.

KurtE, 2023-06-20:
Today I thought I would see if it might also work with the Feather M4, 

First attempt failed as I expected.  I found that the variant feather_m4
did not have either the opencd_scripts nor the debug-scripts like the feather_m0 one did.  
So I copied those over renamed feather_m0.cfg to feather_m4.cfg.
Changed the chip name in it. so the contents looks now like:
```
#
#  Adafruit Feather M0 Express OpenOCD script.
#
#  Copyright (c) 2014-2015 Arduino LLC.  All right reserved.
#
#  This library is free software; you can redistribute it and/or
#  modify it under the terms of the GNU Lesser General Public
#  License as published by the Free Software Foundation; either
#  version 2.1 of the License, or (at your option) any later version.
#
#  This library is distributed in the hope that it will be useful,
#  but WITHOUT ANY WARRANTY; without even the implied warranty of
#  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
#  See the GNU Lesser General Public License for more details.
#
#  You should have received a copy of the GNU Lesser General Public
#  License along with this library; if not, write to the Free Software
#  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
#

source [find interface/cmsis-dap.cfg]

# chip name
set CHIPNAME at91samd51j19
set ENDIAN little

# choose a port here
set telnet_port 0

source [find target/at91samdXX.cfg]

```
The file variant.gdb:
```
#
#  Arduino Zero OpenOCD script.
#
#  Copyright (c) 2014-2015 Arduino LLC. All right reserved.
#
#  This library is free software; you can redistribute it and/or
#  modify it under the terms of the GNU Lesser General Public
#  License as published by the Free Software Foundation; either
#  version 2.1 of the License, or (at your option) any later version.
#
#  This library is distributed in the hope that it will be useful,
#  but WITHOUT ANY WARRANTY; without even the implied warranty of
#  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
#  See the GNU Lesser General Public License for more details.
#
#  You should have received a copy of the GNU Lesser General Public
#  License along with this library; if not, write to the Free Software
#  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
#

# Define 'reset' command
define reset

info reg

break main

# End of 'reset' command
end
[... 38 lines cut ...]
[2023-06-20T18:01:31.352Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```
The Debug Console tab shows:
```
Cortex-Debug: VSCode debugger extension version 1.5.1 git(be7d3c8+dirty). Usaage info: https://github.com/Marus/cortex-debug#usage
Reading symbols from C:\Users\kurte\AppData\Local\Arduino15\packages\adafruit\tools\arm-none-eabi-gcc\9-2019q4\bin\arm-none-eabi-objdump.exe --syms -C -h -w C:/Users\kurte\AppData\Local\Temp\arduino\sketches\97CEBE8D155FC584FC835A5782B36D4B\Blink_hello.ino.elf
Reading symbols from c:/users\kurte\appdata\local\arduino15\packages\adafruit\tools\arm-none-eabi-gcc\9-2019q4\bin\arm-none-eabi-nm.exe --defined-only -S -l -C -p C:/Users\kurte\AppData\Local\Temp\arduino\sketches\97CEBE8D155FC584FC835A5782B36D4B\Blink_hello.ino.elf
Launching GDB: "C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\adafruit\\tools\\arm-none-eabi-gcc\\9-2019q4\\bin\\arm-none-eabi-gdb.exe" -q --interpreter=mi2 "C:/Users\\kurte\\AppData\\Local\\Temp\\arduino\\sketches\\97CEBE8D155FC584FC835A5782B36D4B\\Blink_hello.ino.elf"
    IMPORTANT: Set "showDevDebugOutput": "raw" in "launch.json" to see verbose GDB transactions here. Very helpful to debug issues or report problems
Launching gdb-server: "C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\openocd\\0.10.0-arduino7/bin/openocd" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "c:\\Users\\kurte\\Documents\\Arduino\\Blink_hello" -f "C:/arduino-ide_nightly/resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl" -f "C:\\Users\\kurte\\AppData\\Local\\Arduino15\\packages\\adafruit\\hardware\\samd\\1.7.12/variants/feather_m4/openocd_scripts/feather_m4.cfg"
    Please check TERMINAL tab (gdb-server) for output from C:\Users\kurte\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.10.0-arduino7/bin/openocd
Finished reading symbols from objdump: Time: 41 ms
C:\Users\kurte\AppData\Local\Arduino15\packages\adafruit\tools\arm-none-eabi-gcc\9-2019q4\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
Finished reading symbols from nm: Time: 46 ms
WARNING: Cortex-Debug will deprecate use of GDB version 8 after July 2022. Please upgrade to version 9+
Reading symbols from C:/Users\kurte\AppData\Local\Temp\arduino\sketches\97CEBE8D155FC584FC835A5782B36D4B\Blink_hello.ino.elf...
Failed to launch GDB: Remote connection closed (from target-select extended-remote localhost:50000)
OpenOCD: GDB Server Quit Unexpectedly. See gdb-server output for more details.
"monitor" command not supported by this target.
"monitor" command not supported by this target.
GDB never responded to an interrupt request. Trying to end session anyways
```

Not sure what to try next, as I am not an expert in this, but was curious.

hathach, 2023-06-21:
debug is interesting, cmsis-dap and jlink is fairly popular. I was planning to try out IDE 2.0 for sometime, maybe this is a good chance to get familliar with IDE v2. Let me see if I could get it done quickly.

hathach, 2023-06-22:
PR https://github.com/adafruit/ArduinoCore-samd/pull/346 add debugging support for Adafruit board both M0 and M4. To debug M4, you will need to have openocd at least version 0.11.0. When releasing the new BSP we will include that, for now, you can install "mbed rp2040", it will auto install openocd v0.11.0 (the one come with samd core is only 0.10.0). 

@KurtE would you mind testing the pr out to see if you could debug both m0 and m4 without any modification. Note: you would need an cmsid-dap (daplink) probe such as pi debug probe https://www.adafruit.com/product/5699

mjs513, 2023-06-22:
> PR #346 add debugging support for all Adafruit samd boards both M0 and M4. To debug M4, you will need to have openocd at least version 0.11.0. When releasing the new BSP we will include that, for now, you can install "mbed rp2040", it will auto install openocd v0.11.0 (the one come with samd core is only 0.10.0).
> 
> @KurtE would you mind testing the pr out to see if you could debug both m0 and m4 without any modification. Note: you would need an cmsid-dap (daplink) probe such as pi debug probe https://www.adafruit.com/product/5699

@tiny,
Seems I already had a version 0.11.0 installed so I changed to point the m0 to openocd-0.11.0-arduino2 for the M0 feather board that I have and it worked like a charm.  I did have to change openocd in platforms.txt to:
 
```
debug.server.openocd.path={runtime.tools.openocd-0.11.0-arduino2.path}/bin/openocd
debug.server.openocd.scripts_dir={runtime.tools.openocd-0.11.0-arduino2.path}/share/openocd/scripts/

```
```
Program stopped, probably due to a reset and/or halt issued by debugger
target halted due to debug-request, current mode: Thread 
xPSR: 0x81000000 pc: 0x00000294 msp: 0x20002de0
target halted due to debug-request, current mode: Thread 
xPSR: 0x81000000 pc: 0x00000294 msp: 0x20002de0
Note: automatically using hardware breakpoints for read-only addresses.
Breakpoint 1, loop () at D:\Users\Merli\Documents\Arduino\Blink_m0/Blink_m0.ino:34
34	  digitalWrite(LED_BUILTIN, HIGH);  // turn the LED on (HIGH is the voltage level)
Breakpoint 1, loop () at D:\Users\Merli\Documents\Arduino\Blink_m0/Blink_m0.ino:34
34	  digitalWrite(LED_BUILTIN, HIGH);  // turn the LED on (HIGH is the voltage level)

```

Unfortunately I do not have a M4 board to test with so @KurtE  will have to test on that.

KurtE, 2023-06-23:
Thanks @mjs513 your change to platform.txt to use the 0.11.0 worked like a charm.  So confirmed it worked on my M4 Express board. 

So hopefully this all will get pulled in !

hathach, 2023-06-23:
hmm, it should pick up the later openocd 0.11 if not explicitly specify the version. Let me make it explicit as your suggestion.

hathach, 2023-06-23:
samd 1.7.13 is relesaed, it will take a few hours before it is indexed by arduino. Thank you for bringing this up.

closed 2023-06-23 by hathach